gonzalop/wombatt #3: Decoding issue when connecting tcp:

techytrev, 2023-11-28:
Hi,

First off, thank you for creating this project! This is exactly what I've been looking for.

**Issue:** I have a "PUSR USR-TCP232-304 Serial RS485 to TCP/IP" which I'm pointing wombatt too, to gather information about my EG4 LifePower 4 batteries. The network connection appears to be working, however wombatt is erroring right away with.

`error getting info of ID#2: error decoding ascii string '4'
error getting info of ID#3: error decoding ascii string '' 
error getting info of ID#4: error decoding ascii string '4'`

I've been trying to troubleshoot but unable to determine why wombatt isn't liking the data it's receiving. I'm running the latest build from a pull today.

Here's the command I'm executing, along with a quick telnet test to confirm, the remote address/port are indeed working.  Perhaps my parameters are wrong, but I tried various combo, nothing works so far.

`root@sandbox-ubuntu01:/home/git/wombatt/build/linux/amd64# ./wombatt --log-level="debug" battery-info --device-type="tcp" --address="192.168.2.98:6666" --baud-rate=9600 --battery-type="lifepower4" --protocol="auto" --battery-ids=2,3,4                    
2023/11/28 18:29:54.886670 Opening 192.168.2.98:6666...
2023/11/28 18:29:54.911248 Opening 192.168.2.98:6666...
2023/11/28 18:29:54.936473 Opening 192.168.2.98:6666...
2023/11/28 18:29:54.961757 Opening 192.168.2.98:6666...
2023/11/28 18:29:54.963178 error getting info of ID#2: error decoding ascii string '4'
error getting info of ID#3: error decoding ascii string ''
error getting info of ID#4: error decoding ascii string '4'
root@sandbox-ubuntu01:/home/git/wombatt/build/linux/amd64# telnet 192.168.2.98 6666 Trying 192.168.2.98...
Connected to 192.168.2.98.
Escape character is '^]'.
5t111111Tu$@4񔕔55111111u$D$AՐ4t1u5T1111Tq$@Ր4tU111111Tp$@Ԑ4tu5T111111t$AԐ4t񔕑u5TQ1Q1Qp$AՐ4tuu5TQq$@Ր4tu5tUt$F֐44񐵔u5T񰔐Pv$F֐44UTq$DА7Q3Q1EQQ5EU51QQQ1QQQQ1QQѹ-Q%UAU!UEQQEQ1AU5AU5AU5AUU!UUUU!UUUQ1EE%EEE$AЖ4tu5111111Tv$@ѐ44u5t111111Tp$@4tє511111t$D$AՐ4t1u5T1111Tu$@Ր4tU111111Tq$@Ԑ4t񑱑u5T111111q$A4tu5TQ1Q1Qt$AՐ4tuu5TQt$@Ր4tu5tUv$F֐4t񐵔u5T񰔐ѕPp$F֐4tUTu$DА4U񱔰Q6p^]
telnet> quit
Connection closed.`

My USR-TCP232-304 config:
[image]
[image]

techytrev, 2023-11-28:
FY I - I tried using the 0.0.3 build as well, getting the same issue. 

Thanks in advance for your help.

gonzalop, 2023-12-03:
Can you try with '--protocol=lifepower4' instead? 
Thanks.

gonzalop, 2023-12-03:
wombatt will query the battery IDs provided in the command line. A time out means that either the port wasn't open or there was not a long enough response or a response at all.

I've uploaded a new image that can be retrieved with 'docker pull gonzalomono/wombatt:debug' with the commits from the 'debug' branch I just created. Could try it out and report back? Also, perhaps increase the default read time from 500ms to 2s ("-t 2s").

TIA

techytrev, 2023-12-03:
Thanks @gonzalop. I deployed the new image with your debug branch and the 2s timeout.   Here's the output of the log *attached and inline. I let it go for a minute or so.  Pulling from a single battery with 0 address.  Please let me know if you'd like me to do anything else or require more info.

Config:
  wombatt2:
    image: gonzalomono/wombatt:debug
    privileged: true
    restart: unless-stopped
    command: 
    - battery-info
    - --battery-type=lifepower4
    - --device-type=tcp
    - --protocol=lifepower4
    - --battery-ids=0
    - --address=192.168.2.98:6666
    - --baud-rate=9600
    - --read-timeout=2s

`2023/12/03 23:07:32.612301 Opening 192.168.2.98:6666...
2023/12/03 23:07:32.615175 Connected to 192.168.2.98:6666...
2023/12/03 23:07:32.615249 writing request 42
	hex: 7e323030303441343230303030464441330d
ascii: ~20004A420000FDA3
2023/12/03 23:07:32.637571 got header
	hex: 9076949554b0719784d6903694
ascii: �v��T�q��֐6�
2023/12/03 23:07:32.637620 Opening 192.168.2.98:6666...
2023/12/03 23:07:32.657961 Connected to 192.168.2.98:6666...
2023/12/03 23:07:32.658004 error getting info of ID#0: error decoding ascii string '��'
2023/12/03 23:07:33.405771 Opening 192.168.2.98:6666...
2023/12/03 23:07:33.412833 Connected to 192.168.2.98:6666...
2023/12/03 23:07:33.412857 writing request 42
	hex: 7e323030303441343230303030464441330d
ascii: ~20004A420000FDA3
2023/12/03 23:07:33.441799 got header
	hex: 9036849554b051979486907694
ascii: �6��T�Q����v�
2023/12/03 23:07:33.441952 Opening 192.168.2.98:6666...
2023/12/03 23:07:33.445098 Connected to 192.168.2.98:6666...
2023/12/03 23:07:33.445134 error getting info of ID#0: error decoding ascii string '��'
2023/12/03 23:07:33.926034 Opening 192.168.2.98:6666...
2023/12/03 23:07:33.930939 Connected to 192.168.2.98:6666...
2023/12/03 23:07:33.930967 writing request 42
	hex: 7e323030303441343230303030464441330d
ascii: ~20004A420000FDA3
2023/12/03 23:07:34.152592 got header
	hex: b784d69036948554b070b784d6
ascii: ��֐6��T�p���
2023/12/03 23:07:34.152662 Opening 192.168.2.98:6666...
2023/12/03 23:07:34.153680 Connected to 192.168.2.98:6666...
2023/12/03 23:07:34.153690 error getting info of ID#0: error decoding ascii string 'T�'
2023/12/03 23:07:34.917430 Opening 192.168.2.98:6666...
2023/12/03 23:07:34.918229 Connected to 192.168.2.98:6666...
2023/12/03 23:07:34.918256 writing request 42
	hex: 7e323030303441343230303030464441330d
ascii: ~20004A420000FDA3
2023/12/03 23:07:34.938093 got header
	hex: 9794d1d0069495149041a794d0
ascii: �������A���
2023/12/03 23:07:34.938164 Opening 192.168.2.98:6666...
2023/12/03 23:07:34.939203 Connected to 192.168.2.98:6666...
2023/12/03 23:07:34.939218 error getting info of ID#0: error decoding ascii string '�'
2023/12/03 23:07:35.994170 Opening 192.168.2.98:6666...
2023/12/03 23:07:35.995019 Connected to 192.168.2.98:6666...
2023/12/03 23:07:35.995047 writing request 42
	hex: 7e323030303441343230303030464441330d
ascii: ~20004A420000FDA3
2023/12/03 23:07:36.015077 got header
	hex: 979494c07694952490749794d5
ascii: ����v��$�t���
2023/12/03 23:07:36.015135 Opening 192.168.2.98:6666...
[... 45 lines cut ...]
2023/12/03 23:08:01.144083 error getting info of ID#0: error decoding ascii string '�'
2023/12/03 23:08:26.997430 Opening 192.168.2.98:6666...
2023/12/03 23:08:27.000018 Connected to 192.168.2.98:6666...
2023/12/03 23:08:27.000042 writing request 42
	hex: 7e323030303441343230303030464441330d
ascii: ~20004A420000FDA3
2023/12/03 23:08:27.022107 got header
	hex: b784d59076949534b071979496
ascii: ��Րv��4�q���
2023/12/03 23:08:27.022233 Opening 192.168.2.98:6666...
2023/12/03 23:08:27.025379 Connected to 192.168.2.98:6666...
2023/12/03 23:08:27.025403 error getting info of ID#0: error decoding ascii string '4�'
2023/12/03 23:09:18.462005 Opening 192.168.2.98:6666...
2023/12/03 23:09:18.462803 Connected to 192.168.2.98:6666...
2023/12/03 23:09:18.462828 writing request 42
	hex: 7e323030303441343230303030464441330d
ascii: ~20004A420000FDA3
2023/12/03 23:09:18.484775 got header
	hex: 9036948554b0719794d5d01684
ascii: �6��T�q�����
2023/12/03 23:09:18.484930 Opening 192.168.2.98:6666...
2023/12/03 23:09:18.488078 Connected to 192.168.2.98:6666...
2023/12/03 23:09:18.488104 error getting info of ID#0: error decoding ascii string '��'`

[_solar-wombatt2-1_logs.txt](https://github.com/gonzalop/wombatt/files/13540519/_solar-wombatt2-1_logs.txt)

gonzalop, 2023-12-04:
I'm at a loss here. The response data is supposed to start with a '~' but that's not what we're seeing. The only thing I can think of is changing the baud rate to 19200 (-B 19200). Can you send pictures of how the connections looks like for the ethernet and RS485 wires?

Otherwise, I've asked the person who helped testing the lifepower4 batteries to take a look and see if he can help. Let's give him some time to get my message and see if he's available.

The way I would debug this if I had a lifepower4 would be to attach a computer to both the the RS485 of one or all the batteries and the inverter or device that you use that behaves correctly. I would use the 'forward' command that just sends data back and forth from 2 devices and prints the hex-encoded data

gonzalop, 2023-12-04:
That's great news! I made the same mistake when crimping and that's how I found out that either pins 1-2 or 7-8 return the same data for the EG4LLv2 :).

Thanks a lot and let me know if there's any missing feature that you would like wombatt to provice.

closed 2023-12-04 by gonzalop