gonzalop/wombatt #4: HA/MQTT some warnings/errors with LifePower4 metrics/sensors

techytrev, 2023-12-13:
Hi @gonzalop,

I've had some time to soak wombatt with LifePower4 batteries. Everything is working very well. 
I am seeing a few issues with HA/MQTT (11.2) though, which I thought I'd share with you.

1. Using kWh instead of, W or kW as unit of measurement. "cumulative_discharge_power" & "cumulative_charging_power"

2. Illegal characters for "remaining_capacity" metric.

3. Using '°K' instead of 'K', as unit of measurement. 

See logs below.
`2023-12-12 21:27:55.884 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery2_remaining_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:55.885 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery2_full_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:56.090 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery3_remaining_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:56.091 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery3_full_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:56.226 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery4_remaining_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:56.227 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery4_full_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:56.321 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery5_remaining_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:56.321 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery5_full_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:56.488 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery6_remaining_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:56.489 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery6_full_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:56.593 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery7_remaining_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:56.594 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery7_full_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:56.721 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery8_remaining_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:56.722 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery8_full_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:56.848 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery9_remaining_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:56.872 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery9_full_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:56.917 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery10_remaining_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:56.930 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery10_full_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:57.092 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery11_remaining_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:57.092 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery11_full_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:57.671 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery12_remaining_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:57.671 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery12_full_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:57.807 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery13_remaining_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2023-12-12 21:27:57.808 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery13_full_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
[... cut ...]
2023-12-12 21:27:59.999 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.eg4_pack1_battery_12_cumulative_charging_power (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'kWh' which is not a valid unit for the device class ('power') it is using; expected one of ['W', 'kW']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22`

Thanks

gonzalop, 2023-12-13:
I see. There's a colon in the MQTT topic, and somehow the list of temperature units does not have the degree symbol for K (while it does for C and F!?). I'll take a look in 8-10h and address all the issues. The logs are going to be super useful. Thank you!

techytrev, 2023-12-13:
Thanks for the update.

haha. Yes, I would have never guessed it would want degree symbols for C & F, but not K. ['°C', '°F', 'K'].  But that's what the HA log says. 
If you find you need more info, just let me know.

FYI - here's my wombatt options, just in case it's relevent. 

`    - monitor-batteries 
    - --web-server-address=:8080 
    - --battery-type=lifepower4
    - --device-type=tcp
    - --protocol=lifepower4
    - --battery-id=2,3,4,5,6,7,8,9,10,11,12,13
    - --address=<IP>:6666
    - --baud-rate=9600
    - --mqtt-broker=tcp://<HA_IP>:1883 
    - --mqtt-user=<user>
    - --mqtt-password=<pass>
    - --mqtt-prefix=eg4_pack1`

closed 2023-12-13 by gonzalop

gonzalop, 2023-12-13:
Hopefully that fixes it. I've create v0.0.5 and updated the docker images with these changes.
Thanks!

techytrev, 2023-12-14:
@gonzalop - Thanks again for the awesome support!  I pulled the new docker image.

I can confirm, two of the items (kWh to kW & °K to K) have corrected in HA and the system has indicated as such. It's now complaining old & new data are different, but I think the old units will just phase out over time. No issues here.

I do have 2 issues remaining. I restarted HA a couple times, just to be sure the warnings come back. And they did.

1. Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery11_full_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic

2. Template variable warning: 'dict object' has no attribute 'remaining_capacity' when rendering '{{ value_json.remaining_capacity }}'
Template variable warning: 'dict object' has no attribute 'full_capacity' when rendering '{{ value_json.full_capacity }}'
----------------------------
Logs:
```
Logger: homeassistant.components.mqtt.discovery
Source: components/mqtt/discovery.py:154
Integration: MQTT (documentation, issues)
First occurred: 8:10:45 PM (24 occurrences)
Last logged: 8:10:46 PM
Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery11_full_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery12_remaining_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery12_full_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery13_remaining_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
Received message on illegal discovery topic 'homeassistant/sensor/eg4_pack1_battery13_full_capacity:/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic

Logger: homeassistant.helpers.template
Source: helpers/template.py:2260
First occurred: 8:10:49 PM (24 occurrences)
Last logged: 8:10:50 PM
Template variable warning: 'dict object' has no attribute 'remaining_capacity' when rendering '{{ value_json.remaining_capacity }}'
Template variable warning: 'dict object' has no attribute 'full_capacity' when rendering '{{ value_json.full_capacity }}'
```

Logs for the fixed ones you did, in case you're interested.
```
Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:204
Integration: Sensor (documentation, issues)
First occurred: 8:00:11 PM (96 occurrences)
Last logged: 8:00:12 PM
The unit of sensor.eg4_pack1_battery_13_cell_temp_4 is changing, got multiple {'K', '°K'}, generation of long term statistics will be suppressed unless the unit is stable and matches the unit of already compiled statistics (°K). Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
The unit of sensor.eg4_pack1_battery_13_environment_temp is changing, got multiple {'K', '°K'}, generation of long term statistics will be suppressed unless the unit is stable and matches the unit of already compiled statistics (°K). Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
The unit of sensor.eg4_pack1_battery_13_mosfet_temp is changing, got multiple {'K', '°K'}, generation of long term statistics will be suppressed unless the unit is stable and matches the unit of already compiled statistics (°K). Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
The unit of sensor.eg4_pack1_battery_13_max_cell_temp is changing, got multiple {'K', '°K'}, generation of long term statistics will be suppressed unless the unit is stable and matches the unit of already compiled statistics (°K). Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
The unit of sensor.eg4_pack1_battery_13_min_cell_temp is changing, got multiple {'K', '°K'}, generation of long term statistics will be suppressed unless the unit is stable and matches the unit of already compiled statistics (°K). Go to https://my.home-assistant.io/redirect/developer_statistics to fix this

Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:232
Integration: Sensor (documentation, issues)
First occurred: 8:00:11 PM (27 occurrences)
Last logged: 8:00:12 PM
The unit of sensor.eg4_pack1_battery_11_cumulative_discharge_power (kW) cannot be converted to the unit of previously compiled statistics (kWh). Generation of long term statistics will be suppressed unless the unit changes back to kWh or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
The unit of sensor.eg4_pack1_battery_12_cumulative_charging_power (kW) cannot be converted to the unit of previously compiled statistics (kWh). Generation of long term statistics will be suppressed unless the unit changes back to kWh or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
[... cut ...]
The unit of sensor.eg4_pack1_battery_13_cumulative_charging_power (kW) cannot be converted to the unit of previously compiled statistics (kWh). Generation of long term statistics will be suppressed unless the unit changes back to kWh or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
The unit of sensor.eg4_pack1_battery_13_cumulative_discharge_power (kW) cannot be converted to the unit of previously compiled statistics (kWh). Generation of long term statistics will be suppressed unless the unit changes back to kWh or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
```

Thanks

gonzalop, 2023-12-14:
1. Unfortunately, the discovery topic are set to be retained so they need to be deleted. I use mosquitto and when I've had similar problems, I've stopped the server, deleted the persistent database from disk and started the server again. If that's not an option for you, simply deleting the offending topic(s) will do: `mosquitto_pub -h <mqtttserver> -t the/invalid/topic -u <username> -P <password> -n -r -d`
2. That error should not happen if everything is alright. Maybe HA is still confused due to the typos in the discovery topics or the problem with the kW vs kWh? If not, could you check what you get when running `mosquitto_sub -t homeassistant/sensor/eg4_pack1_battery11_info/state` on any battery? TIA.
3. I assume you've already cleared out the bad statistics using kWh using the /developer-tools/statistics in HA. They should show up with a 'Fix issue' button on the right, and allows you to delete them.

techytrev, 2023-12-14:
Aha, I didn't know about the purging in MQTT or HA statistics. I did a bunch of cleanups and all is looking good, now.

Thanks for your wisdom!