gonzalop/wombatt #6: Add configuration to run wombatt as a docker instance within homeassistant as an add-on

js1tr3, 2024-01-01:
Add configuration files to run wombatt as a docker instance within homeassistant as an add-on

js1tr3, 2024-01-01:
Thanks for building this application! it works great with home-assistant for monitoring my EG4LL batteries 2-4 using the provided Signature Solar USB adapter! I've got it mostly working as a local add-on an hope to contribute the configuration files after testing it for a bit.
downside is it requires protection mode to access the ttyUSB0 Device.

gonzalop, 2024-01-02:
I am running 2 instances outside of the HA VM at the moment: one with monitor-inverters and one with monitor-batteries. Having them running inside of HA and managed by HA would be great.

gonzalop, 2024-01-02:
I use a zwave USB stick plugged in the host and delegated to the HA VM. I
don't remember having to enable any special mode for using the USB stick
other than adding the device to the VM.

commit aa0fa931 (2025-08-16): Add support for running wombat in HA.

wombatt can now run in a docker container managed by Homeassistant.

Closes #6.
closed 2025-08-16 by gonzalop