gonzalop/wombatt #7: Support EG4 6000XP

jsmithtx, 2024-01-25:
Would love to see support for EG4 6000XP. Will download the code and see what I can do to help contribute.

gonzalop, 2024-01-26:
If you can point me to documentation on connection type (RS232, RS485) and some details about the protocol, that would be helpful. For RS485+Modbus, if there's no documentation available, the modbus-read command can be useful to find out what registers are available

gonzalop, 2024-01-26:
Download a release binary of wombatt for you system (windows, linux ARM/ARM64 for a Pi or linux AMD64 for intel), or build wombatt from source, then save the following lines to a file named _format.txt_:
```
u16:state
u16:pv1_voltage:V:0.01
u16:pv2_voltage:V:0.01
u16:pv3_voltage:V:0.01
u16:battery_voltage:V:0.1
u8:soc
u8:soh
u16:internal_fault
u16:pv1_power:W
u16:pv2_power:W
u16:pv3_power:W
u16:charging_power:W
u16:discharging_power:W
```
Now you need to connect the INV485 port (also try the 485A and 485B) to a RS485 to USB converter (i.e., something like [https://www.amazon.com/Industrial-Bidirectional-Resettable-Protection-300bps-3MbpsBaudrateSupport/dp/B0B8DWLT1V/](https://www.amazon.com/Industrial-Bidirectional-Resettable-Protection-300bps-3MbpsBaudrateSupport/dp/B0B8DWLT1V/)) and plug that converter to a computer.

Assuming the wombatt binary and the above file are in the same folder, and that the RS485 to USB converter shows as _/dev/ttyUSB0_ , run the following command:

`./wombatt modbus-read -B 19200 --protocol ModbusRTU -O format.txt -p /dev/ttyUSB0 --id 1 --start 0 --count 12`

If the output makes sense, we're on the right track and I'd need to make a couple of general-purpose changes to support Modbus in the monitor-inverter command, and then add a struct containing all the fields that you deem most relevant (or all?)  to a new struct that will be used to read the modbus data into and display it or send it to MQTT.

jsmithtx, 2024-01-26:
Do you have any thoughts about getting the data via an existing wireless/wired connection? Currently the device is connected to a private wired network (which is how it can send the data to the 'cloud' if I enabled it), so I wonder if there's a tcp command that can get the data.

gonzalop, 2024-01-26:
I don't think that the dongle would expose the port directly, but you could try using 'nmap' to find open ports at the IP address assigned to the dongle and see if any of those ports might be it.

What I would try in this case is a main-in-the-middle approach. For this you need 4 RS485 to USB adapters and a Pi or other computer on which you can run commands and have the 2 adapters connected at the same time. One adapter goes to the inverter, the other adapter goes to the dongle. Then run the `wombatt forward` command and that will show you the data that goes back and forth between the inverter and the dongle. With a bit of luck, that data will use the same modbus protocol documented for the inverter and could be extracted without sending any actual request from the device inspecting the data.

jsmithtx, 2024-02-05:
I did confirm with EG that the WLAN adapter does NOT support getting data.

Where would you connect the RS485 port to that adapter? The only port I know of is the battery comm, but I don't think it would chat about inverter data. I wonder what port I would use...

jsmithtx, 2024-02-05:
Ah yes, I remember that part of the manual. I just need to open up mine and see where these 'meters' are at.

jsmithtx, 2024-02-06:
GOT IT! When I took the adapter cover off, there are 4 wires... blue, red, white, and black

B - black wire
A - white wire

And results...

wombatt.exe modbus-read -B 19200 --protocol ModbusRTU -O format.txt -p COM6 --id 1 --start 0 --count 12
State: 705
Pv1_voltage: 0.01V
Pv2_voltage: 133.63V
Pv3_voltage: 131.1V
Battery_voltage: 1336.7V
Soc: 50%
Soh: 48%
Internal_fault: 13361
Pv1_power: 25443W
Pv2_power: 24929W
Pv3_power: 2838W
Charging_power: 267W
Discharging_power: 0W

jsmithtx, 2024-02-06:
> If the output makes sense, we're on the right track and I'd need to make a couple of general-purpose changes to support Modbus in the monitor-inverter command, and then add a struct containing all the fields that you deem most relevant (or all?)

My one comment on the output is that the actual number values are incorrect. Hopefully it is just an struct mismatch.

gonzalop, 2024-02-06:
> My one comment on the output is that the actual number values are incorrect. Hopefully it is just an struct mismatch.

Right, I was about to say that some values look wrong.
Can you run another command for me?

`./wombatt modbus-read -B 19200 --protocol ModbusRTU -p COM6 --id 1 --start 0 --count 40`

and let me know the output. That should help me writing tests and start fetching the most interesting values from the long list of registers.

It would also be very helpful if you can take a look at the table in https://eg4electronics.com/backend/wp-content/uploads/2023/10/EG4-6000XP-MODBUS-Communication-Protocol.pdf at the table under "3. Register Mapping Table 1) Input register". For the first 40 registers, which will be fetched by the command above, try to provide as many known values as possible. I.e., if you know by other means the value of SOC and SOH, total PV prouction per string, their voltage, charging or discharging power, etc... This will help me narrow down any issues with the struct alignment.

Additional tests like trying ID 2 if you have a second inverter or seeing if lowering the baud rate to 9600 produces different values might also be helpful.

Thanks!

jsmithtx, 2024-02-06:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

jsmithtx, 2024-02-07:
I thought I would get clever and write the hex int values out to see if I could correlate them, but it seems to have made it even worse. Here are the first 20:

```No | HexBA  | BANo  | ANo | BNo
00 | 0xC102 | 49410 | 002 | 193
01 | 0x0100 | 00256 | 000 | 001
02 | 0x3334 | 13108 | 052 | 051
03 | 0x3633 | 13875 | 051 | 054
04 | 0x3734 | 14132 | 052 | 055
05 | 0x3032 | 12338 | 050 | 048
06 | 0x3134 | 12596 | 052 | 049
07 | 0x6363 | 25443 | 099 | 099
08 | 0x6161 | 24929 | 097 | 097
09 | 0x160B | 05643 | 011 | 022
10 | 0x0B01 | 02817 | 001 | 011
11 | 0x0000 | 00000 | 000 | 000
12 | 0x1802 | 06146 | 002 | 024
13 | 0x0701 | 01793 | 001 | 007
14 | 0x1425 | 05157 | 037 | 020
15 | 0x0100 | 00256 | 000 | 001
16 | 0x0100 | 00256 | 000 | 001
17 | 0x0000 | 00000 | 000 | 000
18 | 0x0000 | 00000 | 000 | 000
19 | 0x2600 | 09728 | 000 | 038

jsmithtx, 2024-02-08:
Got it!

The command you sent is reading the *holding* registers, not the *input* registers. I'm not sure what command to issue for those, but I found a modbus c# library and read the input registers and got data that is actually meaningful (I was able to match the serial number of register 115, length 5) (although the bytes are reversed).

jsmithtx, 2024-02-09:
Ok, good grief.. sorry to spam you. I *finally* got the numbers to match. I should *not* reverse the hex values even though I must reverse for the serial number -- no idea why! 

This reading is taken at *night*-- and I think a lot of 0s because the inverter is UTC so new day.

```
No   |  AB-Hex  |  AB-Int |  A-Int  |  B-Int
000  |  0x0040  |  00064  |   000   |   064 // correct, 0x40 is "night" mode (battery only); Day will switch to C0 (second hex)
001  |  0x0000  |  00000  |   000   |   000
002  |  0x0000  |  00000  |   000   |   000
003  |  0x0F49  |  03913  |   015   |   073 // there is no PV3, I can only assume that it just makes stuff up
004  |  0x020D  |  00525  |   002   |   013 // correct, battery voltage is 52.5v (AB-Int)
005  |  0x0057  |  00087  |   000   |   087 // correct, SOC is 87% (B-Int)
006  |  0x0000  |  00000  |   000   |   000
007  |  0x0000  |  00000  |   000   |   000
008  |  0x0000  |  00000  |   000   |   000
009  |  0x0000  |  00000  |   000   |   000
010  |  0x0000  |  00000  |   000   |   000
011  |  0x009A  |  00154  |   000   |   154 // correct, discharge is around ~150w (AB-Int)
012  |  0x0000  |  00000  |   000   |   000
013  |  0x0000  |  00000  |   000   |   000
014  |  0x1801  |  06145  |   024   |   001
015  |  0x0000  |  00000  |   000   |   000
016  |  0x0000  |  00000  |   000   |   000
017  |  0x0000  |  00000  |   000   |   000
018  |  0x005A  |  00090  |   000   |   090
019  |  0x0000  |  00000  |   000   |   000
020  |  0x095F  |  02399  |   009   |   095
021  |  0x2740  |  10048  |   039   |   064
022  |  0x5052  |  20562  |   080   |   082
023  |  0x176D  |  05997  |   023   |   109
024  |  0x0092  |  00146  |   000   |   146
025  |  0x00B6  |  00182  |   000   |   182
026  |  0x0000  |  00000  |   000   |   000
027  |  0x0000  |  00000  |   000   |   000
028  |  0x0000  |  00000  |   000   |   000
029  |  0x0000  |  00000  |   000   |   000
030  |  0x0000  |  00000  |   000   |   000
031  |  0x0000  |  00000  |   000   |   000
032  |  0x0000  |  00000  |   000   |   000
033  |  0x0000  |  00000  |   000   |   000
034  |  0x0003  |  00003  |   000   |   003
035  |  0x0002  |  00002  |   000   |   002
036  |  0x0000  |  00000  |   000   |   000
037  |  0x0000  |  00000  |   000   |   000
038  |  0x0F51  |  03921  |   015   |   081
039  |  0x0E88  |  03720  |   014   |   136
...
096  |  0x0004  |  00004  |   000   |   004  //4 is correct
097  |  0x0190  |  00400  |   001   |   144  //400 correct

gonzalop, 2024-02-16:
The first reply that you posted, starting with bytes "02 c1" seems like some kind of error response because the second byte has the highest bit set to 1.

You are correct and I should be using the 'Read Input Register' (0x04) command instead of the 'Read Holding Register' (0x03) command. That's an easy fix.
 
You're getting somewhere in this last message. How did you get that data? Specifically I'd like to know *all* the bytes sent to request it and *all* the bytes that were received, not just the register data bytes.

jsmithtx, 2024-02-20:
Interesting note on the 0xC1, it should be C0. I'll have to pull a fresh check tomorrow when charging and see if it goes back to C0. I think it will; I had the battery comm messed up. An earlier communication had it as "00 C0".

I got the data the same way you are getting it, just using the **input** rather than holding registers. If I pull the "holding" ones I get the exact match with what wombatt returns.

I am using FluentModbus (nuget C# library); 

```
using var client = new ModbusRtuClient()
{
    BaudRate = 19200,
    StopBits = StopBits.One,
    Parity = Parity.None
};
client.Connect("COM6", ModbusEndianness.LittleEndian);
var registers = client.ReadInputRegisters(0x01, 0, 40).ToArray();
```

jsmithtx, 2024-02-20:
Ok, to follow up from last night, my holding numbers match exactly with your wombatt command. So whatever you are doing, if you change it to input registers (0x04), you'll probably get the same data below.

I dumped the data from 138 registers (and snipped my serial number part out). For the data I could identify and matched correctly, I put a comment on the line with name=value. Some of the lines make no sense (marked with ?) -- weird values that seem wild. Both "L1" and "L2" eps output voltage should be 120/121 volts, but didn't find that below.

```
Data at 2024-02-20T14:42:56:
00 C0 09 CA 00 00 0F 80 02 14 00 58 00 00 01 54 00 00 00 00 00 EF 00 00 00 00 00 00 18 01 00 00 00 00 00 00 00 64 00 00 09 78 27 40 20 02 17 71 00 26 00 26 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 0B 00 0C 00 00 00 00 0F 78 0E AF 00 E3 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 93 00 00 00 50 00 00 00 79 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1F 00 2E 00 31 00 2B 00 00 36 CA 00 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 3A 98 75 30 02 30 00 00 00 00 00 00 C0 00 00 00 00 00 12 00 AB BF 00 00 60 1C 04 A5 00 03 00 04 01 90 01 5E 00 00 00 00 0D 09 0C FD 00 DC 00 6E 02 16 00 01 02 13 00 00 00 00 00 00 00 00 00 00 01 04 00 00 .. .. .. .. .. .. .. .. .. .. 07 CA 00 00 00 00 36 E0 02 00 7C C1 00 14 04 AF 04 BA 00 10 00 15 00 10 00 18 00 02 00 0B 00 10 00 00 00 6C

No   |  AB-Hex  |  AB-Int |  A-Int  |  B-Int
000  |  0x00C0  |  00192  |   000   |   192  // state = PV off-grid (0xC0)
001  |  0x09CA  |  02506  |   009   |   202  // vpv1 = 250.6v
002  |  0x0000  |  00000  |   000   |   000  // vpv2 = 0
003  |  0x0F80  |  03968  |   015   |   128  // vpv3 = ???? (does not exist)
004  |  0x0214  |  00532  |   002   |   020  // vbat = 53.2v
005  |  0x0058  |  00088  |   000   |   088  // SOC = 88%
006  |  0x0000  |  00000  |   000   |   000  // internal fault code = 0 (none)
007  |  0x0154  |  00340  |   001   |   084  // Ppv1 = 340w
008  |  0x0000  |  00000  |   000   |   000  // Ppv2 = 0
009  |  0x0000  |  00000  |   000   |   000  // Ppv3
010  |  0x00EF  |  00239  |   000   |   239  // Pcharge = 239w
011  |  0x0000  |  00000  |   000   |   000  // Pdisch = 0
012  |  0x0000  |  00000  |   000   |   000  // VacR
013  |  0x0000  |  00000  |   000   |   000  // VacS
014  |  0x1801  |  06145  |   024   |   001  // VacT  ???
015  |  0x0000  |  00000  |   000   |   000  // Fac
016  |  0x0000  |  00000  |   000   |   000  // Pinv
017  |  0x0000  |  00000  |   000   |   000  // Prec
018  |  0x0064  |  00100  |   000   |   100  // IinvRMS ??
019  |  0x0000  |  00000  |   000   |   000  // PF
020  |  0x0978  |  02424  |   009   |   120  // VepsR  ???
021  |  0x2740  |  10048  |   039   |   064  // VepsS ???
022  |  0x2002  |  08194  |   032   |   002  // VepsT  ???
023  |  0x1771  |  06001  |   023   |   113  // Feps = 60.01Hz
024  |  0x0026  |  00038  |   000   |   038  // Peps 
025  |  0x0026  |  00038  |   000   |   038  // Seps 
026  |  0x0000  |  00000  |   000   |   000  // Ptogrid = 0
027  |  0x0000  |  00000  |   000   |   000  // Ptouser = 0
028  |  0x0002  |  00002  |   000   |   002  // Epv1_day = 0.2kWh
029  |  0x0000  |  00000  |   000   |   000  // Epv2_day = 0
030  |  0x0000  |  00000  |   000   |   000  // Epv3_day = 0
031  |  0x0000  |  00000  |   000   |   000  // Einv_day = 0
032  |  0x0000  |  00000  |   000   |   000  // Erec_day = 0
033  |  0x0000  |  00000  |   000   |   000  // Echg_day = 0
034  |  0x000B  |  00011  |   000   |   011  // Edischg_day = 1.1kWh
035  |  0x000C  |  00012  |   000   |   012  // Eeps_day = 1.2kWh
036  |  0x0000  |  00000  |   000   |   000  // Etogrid_day = 0
037  |  0x0000  |  00000  |   000   |   000  // Etouser_day = 0
038  |  0x0F78  |  03960  |   015   |   120  // Vbus1  ??
039  |  0x0EAF  |  03759  |   014   |   175  // Vbus2  ??
040  |  0x00E3  |  00227  |   000   |   227  // Epv1_all L=22.7kWh
041  |  0x0000  |  00000  |   000   |   000
042  |  0x0000  |  00000  |   000   |   000
043  |  0x0000  |  00000  |   000   |   000
044  |  0x0000  |  00000  |   000   |   000
045  |  0x0000  |  00000  |   000   |   000
046  |  0x0000  |  00000  |   000   |   000
047  |  0x0000  |  00000  |   000   |   000
048  |  0x0000  |  00000  |   000   |   000
049  |  0x0000  |  00000  |   000   |   000
050  |  0x0093  |  00147  |   000   |   147  // Echg_all L=14.7kWh
051  |  0x0000  |  00000  |   000   |   000
052  |  0x0050  |  00080  |   000   |   080
053  |  0x0000  |  00000  |   000   |   000
054  |  0x0079  |  00121  |   000   |   121  // Eeps_all L=12.1kWh
055  |  0x0000  |  00000  |   000   |   000
056  |  0x0000  |  00000  |   000   |   000
057  |  0x0000  |  00000  |   000   |   000
058  |  0x0000  |  00000  |   000   |   000
059  |  0x0000  |  00000  |   000   |   000
060  |  0x0000  |  00000  |   000   |   000
[... 48 lines cut ...]
109  |  0x0000  |  00000  |   000   |   000
110  |  0x0000  |  00000  |   000   |   000
111  |  0x0000  |  00000  |   000   |   000
112  |  0x0000  |  00000  |   000   |   000
113  |  0x0104  |  00260  |   001   |   004
114  |  0x0000  |  00000  |   000   |   000
..snip..
120  |  0x07CA  |  01994  |   007   |   202  // VBusP  ??
121  |  0x0000  |  00000  |   000   |   000
122  |  0x0000  |  00000  |   000   |   000
123  |  0x36E0  |  14048  |   054   |   224  // no generator; numbers odd
124  |  0x0200  |  00512  |   002   |   000
125  |  0x7CC1  |  31937  |   124   |   193
[... cut ...]
129  |  0x0010  |  00016  |   000   |   016  // Peps_L1N=16w
130  |  0x0015  |  00021  |   000   |   021  // Peps_L2N=21w
131  |  0x0010  |  00016  |   000   |   016  // Seps_L1N=16VA
132  |  0x0018  |  00024  |   000   |   024  // Seps_L2N=24VA
133  |  0x0002  |  00002  |   000   |   002  // EepsL1N_day=0.2kWh
134  |  0x000B  |  00011  |   000   |   011  // EepsL2N_day=1.1kWh
135  |  0x0010  |  00016  |   000   |   016 
136  |  0x0000  |  00000  |   000   |   000
137  |  0x006C  |  00108  |   000   |   108

gonzalop, 2024-05-28:
Note to self: use the changes in commit #846d4476f30dc742b20252bd459f641a57dbdcc4 to add a new protocol for monitor-inverters

commit cd0821bf (2025-08-17): Add support for the EG4 6000XP inverter

Closes #7.
commit 2a93c917 (2025-08-17): Add support for the EG4 6000XP inverter

Closes #7.
closed 2025-08-17 by gonzalop