gonzalop/wombatt #13: No cell alarms are shown for lifepower4 batteries in "battery-info" command.

sokhapkin, 2024-05-28:
For lifepower4 batteries alarms are not shown properly in "battery-info" - for each cell alarm status is always 0, even if "voltage event code: cell OV alarm". FET status code never shows "Charge current limiter is ON" even if eg4 monitoring site shows decreased "BMS limit charge" and "BMS Charge: Forbidden".

gonzalop, 2024-05-28:
Could you try running wombatt using the 'debug' branch (https://github.com/gonzalop/wombatt/tree/debug) when any of the alarms is on and the output is wrong and paste here the entire output?
Thanks.

sokhapkin, 2024-05-28:
Here it is. How about to add "-d" (--debug) option to the main branch?
$ ./wombatt battery-info -p /dev/ttyUSB0 --bms-type=lifepower4 --battery-id=2
2024/05/28 19:53:57.852827 Writing: 7e323030323441343230303030464441310d
2024/05/28 19:53:58.140438 Reading: 1101100db60dba0db00db20dae0db70daf0daf0db50dae0db80dbb0db10dbb0db60db7040bf50bf50bf50bf50bff0bffff1f15ed2706270600120f006400630dbb0dae000d0bf50bf500026e4200023ac40001501600012e52000000e7000000f70b8e124cd0
2024/05/28 19:53:58.140619 Writing: 7e323030323441343430303030464439460d
2024/05/28 19:53:58.284288 Reading: 110110000000000000000000000000000000000400000000000000000900110000000003010000000000ed
Battery #2
===========
alarm flag: 0x10, unread alarms
cell 1 voltage: 3.51V
cell 2 voltage: 3.514V
cell 3 voltage: 3.504V
cell 4 voltage: 3.506V
cell 5 voltage: 3.502V
cell 6 voltage: 3.511V
cell 7 voltage: 3.503V
cell 8 voltage: 3.503V
cell 9 voltage: 3.509V
cell 10 voltage: 3.502V
cell 11 voltage: 3.512V
cell 12 voltage: 3.515V
cell 13 voltage: 3.505V
cell 14 voltage: 3.515V
cell 15 voltage: 3.51V
cell 16 voltage: 3.511V
cell temp 1: 306.1K
cell temp 2: 306.1K
cell temp 3: 306.1K
cell temp 4: 306.1K
environment temp: 307.1K
mosfet temp: 307.1K
pack current: -2.25A
pack voltage: 56.13V
remaining capacity: 99.9Ah
full capacity: 99.9Ah
cycle counts: 18
user defined: 15
soc: 100%
soh: 99%
max cell voltage: 3.515V
min cell voltage: 3.502V
diff cell voltage: 0.013V
max cell temp: 306.1K
min cell temp: 306.1K
cumulative charging capacity: 1592.98A
cumulative discharge capacity: 1461.16A
cumulative charging power: 86.038kW
cumulative discharge power: 77.394kW
cumulative charging time: 231h
cumulative discharge time: 247h
cumulative charging times: 2958h
cumulative discharge times: 4684h
alarm flag: 0x10, unread alarms
cell 1 alarm status: 0
cell 2 alarm status: 0
cell 3 alarm status: 0
cell 4 alarm status: 0
cell 5 alarm status: 0
cell 6 alarm status: 0
cell 7 alarm status: 0
cell 8 alarm status: 0
cell 9 alarm status: 0
cell 10 alarm status: 0
cell 11 alarm status: 0
cell 12 alarm status: 0
cell 13 alarm status: 0
cell 14 alarm status: 0
cell 15 alarm status: 0
cell 16 alarm status: 0
cell temp 1: 0
cell temp 2: 0
cell temp 3: 0
cell temp 4: 0
environment temp alarm status: 0
mosfet temp alarm status: 0
pack current alarm status: 0
pack voltage alarm status: 0
user defined: 9
balance event code: 0
voltage event code: pack OV alarm, cell OV alarm
temperature event code: 0
current event code: 0
remaining capacity alarm: 0
fet status code: charge MOS, discharge MOS
system status code: discharging
balance status code: 0

gonzalop, 2024-05-29:
> Here it is. How about to add "-d" (--debug) option to the main branch?

I will make the '-l' option functional soon, probably going with Go's own slog.

> [...]
Looking now at what's wrong with the following 2 lines:
```
voltage event code: pack OV alarm, cell OV alarm
fet status code: charge MOS, discharge MOS
```

gonzalop, 2024-05-29:
Mmm, so the values received from the battery are what is being displayed. See the '11' after '0900' in the second 'Reading' line.

The alarm flag is saying there are unread alarms. Check out https://diysolarforum.com/threads/eg4-life-power4-battery-won%E2%80%99t-wake-up.81431/#post-1050909  and https://diysolarforum.com/threads/anyone-else-with-a-recent-eg4-48v-lifepower4-battery-purchase-getting-a-charging-mos-fault-alarm-that-cant-be-reset.80167/ for how to reset those values.

sokhapkin, 2024-05-29:
Here is what I see frequently - "alarm flag: 0x10", which means wrong (not conforming to the documentation) value. Which fools me to believe that the EG4 documentation is interpreted wrong (or the documentation is wrong). 

fet status code: charge MOS, discharge MOS

is 100% good and means that both charge and discharge FETs are open. The problem is why with the condition "voltage event code: pack OV alarm, cell OV alarm" no alarms are shown for the individual cells. But I believe the question should be addressed to EG4, not to you.

gonzalop, 2024-05-29:
The doc I used as reference is in the repo and I believe I followed what's there, although you've found some mistakes.

Perhaps those values from the EG4 monitoring web site come from the inverters?

gonzalop, 2024-05-29:
Closing since there is nothing for us to do unless we want to pretend the values received from the batteries are not there.

closed 2024-05-29 by gonzalop