gonzalop/wombatt #16: RPI 2b Compilation Issues

vgthecoder, 2024-07-18:
I get this error when trying to do `make`
```
ERRO Running error: 1 error occurred:
	* can't run linter goanalysis_metalinter: goanalysis_metalinter: shift: package "modbus" (isInitialPkg: true, needAnalyzeSource: true): runtime error: invalid memory address or nil pointer dereference
	make: *** [Makefile:101: golangci-lint-run] Error 3
	```

gonzalop, 2024-07-18:
I haven't see that one before. It could be an old golangci-list.

Can you run `go install github.com/golangci/golangci-lint/cmd/golangci-lint@v1.59.1`? I'll update the Makefile help text to this version if it works for you.

Thanks!

vgthecoder, 2024-07-19:
Thank you, That worked perfectly!

closed 2024-07-19 by vgthecoder