gonzalop/wombatt #20: Errors when Using Lifepower4 EG4LLv2 Batteries

vgthecoder, 2024-09-07:
I get these errors when I try using a EG4LL (lifepower4 
./wombatt battery-info --address /dev/ttyUSB0 --battery-id 16 --protocol lifepower4
time=2024-09-07T17:22:45.900-04:00 level=INFO msg="error getting info of ID#16: error decoding ascii string '\xff\xff'"

./wombatt battery-info --address /dev/ttyUSB0 --battery-id 16  --bms-type EG4LLv2
time=2024-09-07T17:25:59.871-04:00 level=INFO msg="error getting info of ID#16: invalid function code: ff"

./wombatt battery-info --address /dev/ttyUSB0 --battery-id 16  --bms-type lifepower4
time=2024-09-07T17:26:50.736-04:00 level=INFO msg="error getting info of ID#16: error decoding ascii string '\xff\xff'"

gonzalop, 2024-09-08:
I've created a 'debug' branch and added a debug line for the header read from the battery in it. Can you checkout https://github.com/gonzalop/wombatt/tree/debug and run it like 'wombatt -l debug battery-info ...' ? What 's the header data printed to the output?

vgthecoder, 2024-09-09:
I am currently not near the batteries, but when I get a chance, I will do that and let you know how it goes.

vgthecoder, 2024-09-20:
## How It Went
All the commands I ran are listed below.
## Attempt 1
### Command:
./wombatt -l debug battery-info --address /dev/ttyUSB0 --battery-id 0 --protocol lifepower4 --bms-type EG4LLv2
### Result:
`level=DEBUG msg="lifepower4 header" data=7e323030f57e323030f57e3230`
`level=INFO msg="error getting info of ID#0: error decoding ascii string '\xf5~20'"`
## Attempt 2
### Command: 
./wombatt -l debug battery-info --address /dev/ttyUSB0 --battery-id 0 --protocol auto --bms-type EG4LLv2
### Result: 
`level=INFO msg="error getting info of ID#0: invalid function code: 32"`
## Attempt 3
### Command:
./wombatt -l debug battery-info --address /dev/ttyUSB0 --battery-id 0
### Result:
`level=INFO msg="error getting info of ID#0: invalid function code: 32"`

gonzalop, 2024-09-21:
The response looks like what a lifepower4 would return (starts with '7e') but then 'f5' should not be there because the lifepower4 that pair of bytes is supposed to be readable as numbers that represent a hexadecimal number (i.e. "31 30" -> ASCII for '1' and '0' -> 10h -> 16), which is why parsing the data fails because 'f5' is not expected.

The subject of the issue says lifepower4 and EG4LLv2 but you seem to have an EG4-LL or EG4-LL-S which uses a different protocol. Can you confirm this? If yes, I have the documentation for EG4-LL modbus and I can write something up that adds support for this variant if you are willing to test it.

vgthecoder, 2024-09-22:
Yes I am willing to test. The unit has the text written on it: EG4LL (lifepower), so I'd assume it is a EG4LL by lifepower. I'd love it if you made something to make it work. Thank you :)

vgthecoder, 2026-01-20:
Okay, will do