gonzalop/wombatt #30: lifepower4 bms_type and protocol are confusing

dulitz, 2025-01-01:
Thanks for putting this together. I have some documentation/interface requests.

I have a Lifepower4 version 2 battery. They're clearly made by EG4 but they do not have "EG4-LL" or "LL" branding on them anywhere.

These batteries have protocol selection DIP switches. In order to communicate with BMS TOOLS from EG4, you place switch 1 in the OFF (down) position and 2-6 in the ON (up) position as shown below.
[image]

It seemed reasonable to try `--bms-type=lifepower4` but it did not work, always timing out. `--bms_type=EG4LLv2` must be used instead. Even with that, `--protocol=lifepower4` does not work. `--protocol=auto` or `--protocol=ModbusRTU` must be used instead.

With all ID DIP switches in the ON (up) position, the battery responds at ID 64.

It would be helpful for these non-obvious things to be documented somewhere, I think. :)

Thanks again; this is very useful.

gonzalop, 2025-01-02:
Oh, interesting. So the "new" Lifepower V2 speak the EG4LLv2 protocol!

I could add an alias for BMS type EG4LLv2, say "lifepower4v2", that should be easy to find in the command-line help. In addition, I should add the command-line BMS type next to each battery type listed as supported in https://github.com/gonzalop/wombatt.

Does that SGTY?

Thanks!

dulitz, 2025-01-02:
Yes, it's good that they don't have Yet Another Protocol. I think adding an alias for the BMS type is great and I like your idea for adding the BMS type next to each battery type on the README.

The only thing I'd add is that, because the protocol is selectable, the user should make sure protocol DIP switch 1 is off and the other protocol DIP switches are on, as shown in my image above. Or at least if a different protocol is selected by DIP switch, the EG4LLv2 protocol may not work (I haven't checked).

On a different note, it wasn't quite clear to me why there is a --protocol option in addition to --bms-type. When would someone use a protocol other than "auto"?

Thanks again, you made it easy to figure out what was going on with this BMS.

gonzalop, 2025-01-02:
Ack. I'll add the alias and update the README including a note about the DIP switch for lifepower4v2.

The --protocol option combined with --bms-type and --device-type define the transport used for communication and the protocol used to talk to the BMS. Normally selecting --bms-type and leaving the other two set to 'auto' works fine. the --protocol and --device-type help here. For instance, in my setup, wombatt connects to a device over TCP which bridges the connection to my EG4v2LL batteries. The data over the TCP connection uses Modbus RTU, not what would be the default for a TCP device type (ModbusTCP). This means that I have to specify **-T tcp --protocol ModbusRTU** since the default bms-type is already correct for me.

closed 2025-01-05 by gonzalop