gonzalop/wombatt #31: Support EG4 12000XP

ddxv, 2025-01-06:
How can I go about checking if this works?

gonzalop, 2025-01-06:
I'm sure it won't work. The first thing we w would need is to get the manufacturer's documentation of the communications protocol used. With that, it would take a couple of hours to add the code needed and then you could test it. If you find the doc in some forum or by emailing the distributor/manufacturer, please, attach it to this issue and I'll do my part.

ddxv, 2025-01-06:
> The first thing we would need is to get the manufacturer's documentation of the communications protocol used. With that, it would take a couple of hours to add the code needed and then you could test it. If you find the doc in some forum or by emailing the distributor/manufacturer, please, attach it to this issue and I'll do my part.

Sure, just emailed support, but a generic email. Will try calling if that doesn't result in any info.